jellewie/Arduino-CocktailMachine #18: The ultimate dispenser project

jellewie, 2024-02-07:
The PUMP dispenser has the problem of that it is not possible for coke and such, since those auto expand and we have a open connection to the outside, so the drink will spill out and run flat.

The SHOTDISPENSER has the issue that it can not do plastic bottles, and even with glass ones it sometimes does not fill the reservoir.

This ticket is there to discuss and track new ideas and types that could overcome these issues. and would be able to dispense coke

jellewie, 2024-02-07:
I have designed a new type of bottle nozzle, this has 2 tubes. One for air to go in, and one for fluid to go out. This would fit on most bottles. [Not in this REPO yet]

Selbie has ordered some valves for testing, it we could mount these below the new nozzle on both sides. If we can mount these on both tubes, this could solve the issue (If the valves are save and work good)

Selbiecookie, 2024-02-07:
Ordered 3 types of solenoid valves (all 12VDC versions), will check out if those can work as a complete replacement for the current dispensers:
- https://aliexpress.com/item/1005005779624351.html
- https://aliexpress.com/item/1005005469057504.html
- https://aliexpress.com/item/1005002876609358.html

Have to check out how they work mechanically, since it has to be food safe, and we do not want any lubricant or other potential harmful substances to enter the liquid that goes through it.

jellewie, 2024-02-08:
Maybe we can re-use the mounting from the Rack assembly, This would make it easy to quick-mount the bottles (bottom and top) and we wouldn't need to design that.

On the other hand, we would still require the valves to be connected to the nozzle, so the unit wouldn't be easily taken off. Unless we design something more for that too? Maybe an even more ideal one is to give the dispenser a little board to send the ID to the main board (same as PNP machines do it) so the machine knows which bottle is connected where (Which bottle ID is which drink settings). And we do not need to manually track that? But this would require quite some engineering in the code, but also in CAD so in the end we might not use them anyway.

jellewie, 2025-03-24:
V2 is here \0/

closed 2025-03-24 by jellewie