jellewie/Arduino-CocktailMachine #23: Update/Redesign PCB Schematic + Layout

Selbiecookie, 2024-02-07:
Update/Redesign PCB Schematic + Layout:
- Re-route traces/connections
- Add more pump circuits
- Add input for sensor for glass detection

jellewie, 2024-02-08:
Can we rename the "pump" in the schematic to a more universal one like for example "switching output"?

jellewie, 2024-02-08:
#18 The idea has come up that it could be nice if the mainboard was aware which DispenerID is connected to which slot. This would also require an input multiplexer. 

I would like to use the CD74HC4067SM96 (Its in my EasyEda lib) to use as the input multiplexer. I believed this one also had the option to be used as an output multiplexer, but we need to test that.

with 4 select wires, 1 data one. and then 2 for the enable pin on 2 units, we can have 16 inputs and 16 outputs (When not used at the same time, which is what we do)

jellewie, 2025-03-24:
Work in progress, we need 3 new PCB types

jellewie, 2025-04-18:
PCB's desiged and underway for testing